socketcan-rs/socketcan-rs #6: non blocking write_frame

wijayaadhisurya, 2017-05-17:
I see that write_frame and write_frame_insists serve different purpose. write_frame_insists is supposed to be blocking while write_frame isn't. So, I tried to use write_frame with expectation that it won't block and return immediately, practically just adding can frame to the tx_queue. However, I observed that between each frame, there is a delay about 1 ms. When I read socketcan app in C/C++, I found this little piece of code:

`fcntl(soc, F_SETFL, O_NONBLOCK);`

I forked this repo without informing you first to particularly add that line of code, because I need it right away in my project. So, what do you think to handle this issue? How do we expose this configuration? adding set_nonblocking method to CANSocket struct? Add new parameter open method?

FYI, I am using v1.6.

mbr, 2017-05-19:
AFAIK messages can fail even when sending a single blocking message, due to
external influences on the bus (the CAN interface getting starved by
higher-priority bus participants). The `_insists` variant was meant to
handle those errors, but still be blocking.

SocketCAN does not handle non-blocking IO at the moment -- you can tell
from the lack of a way to enable non-blocking modes =). The documentation
could be clearer on this, I admit.

Just setting the `fnctl` is not the answer here. At the time the library
was written, there was no support for non-blocking IO for any kind of
socket in Rust. Without an official API to mirror, I decided to just not
implement it and wait.

Seems the time may finally be ripe to add this feature. But I will have to
look into it more carefully.

mbr, 2017-05-19:
Addtionally, I've been holding off releasing 2.0 for quite some time now.
The new major version seems a good fit for these kind of API changes.

wijayaadhisurya, 2017-05-23:
How can I help you with this? 
Should I learn all about UNIX sockets? 
And, what is new major Rust version? Async I/O? futures?

mbr, 2017-06-28:
Closing in favor of #8.

closed 2017-06-28 by mbr