socketcan-rs/socketcan-rs #8: Using socketcan with asynchronous loop

RouquinBlanc, 2017-06-27:
Hi Marc,

For a quick project I have been playing with socketcan, but all my other IO (mainly TCP) were using [Metal IO](https://github.com/carllerche/mio); as a first run I'm using 2 different threads, but in a short future it would be nice to read/write to the CAN socket directly from the async loop!

2 things are missing at the moment, which I added in a fork [in this fork](https://github.com/RouquinBlanc/socketcan-rs/tree/release/1.6):
- access to the internal file descriptor
- API to set as non-blocking. (Directly related to #6)

Then I wrote this small [miocan](https://github.com/RouquinBlanc/miocan-rs) adapter to make the glue between `socketcan 1.6` and `mio 0.6.9`, plus an [example of usage](https://github.com/RouquinBlanc/miocan-example-rs) which works nicely on read.

I don't know yet what would be the best course (integration in socketcan? in mio? 3rd repo as now?), but at least it's pretty cool to bootstrap that work!

Please comment when you have time.
Cheers!

mbr, 2017-06-28:
Have a look at the v1.7.0. I've cleaned up the code and incorporated all the features you wrote I believe.

Please test it with your code (exta credit for adding tests or manually testing setting the device from nonblocking to blocking mode). If it works correctly, I'll release 1.7.0 with those features added.

As for `mio` -- not before 2.0.0, if ever. I need to dig way deeper into `mio` before I can make that decision, sorry. Best to keep it as a separate crate for now, although I can imagine it becoming `feature` at some point.

RouquinBlanc, 2017-06-28:
This looks great.

Tested with my test code on linux with `mio`, works well. And compile correctly on MacOS!

Thanks a lot!

mbr, 2017-06-29:
Thanks. Released as v1.7.0.

closed 2017-06-29 by mbr