socketcan-rs/socketcan-rs #11: Unmaintained?

twaddington, 2019-07-02:
This crate appears to be unmaintained. Any plans to continue supporting?

cwoodall, 2019-11-01:
this + one that can support windows can under one interface (think pycan) would be useful for me.

mbr, 2020-01-25:
I'm officially looking for maintainers to help out, as of now. Please feel free to mail me if you're still interested (my email is in the commits). (/cc @marcelbuesing )

fpagliughi, 2021-02-20:
Hey All. I've been in touch with @mbr and have volunteered to get v2.0 working and tested, then he can merge and push to crates.io.

My goals are:

1. Get the timestamp code tested.
2. Get Netlink code updated and stable.
3. Possibly make the Netlink stuff a (default?) feature that can be compiled out. (Especially if the external library isn't fully stable).
4. Finish updating Edition 2018 (if not complete)
5. Update the dependencies
6. Make the errors conform to std::error::Error
7. Add some examples
8. Maybe some additional unit or integration tests?

I'll be working out of my fork for the next few weeks, if anyone has any of this done, we can work on it over there, then PR back here when it's all working.
https://github.com/fpagliughi/socketcan-rs

slyshykO, 2021-02-21:
@fpagliughi It would be nice to add the ability to directly specify the type of CanFrame identifier as asked in #15.

marcelbuesing, 2021-02-21:
I think it would be great to implement the [embedded-can](https://github.com/timokroeger/embedded-can) traits. Main benefit being that one could easily switch between `socketcan` and embedded implementations. E.g. start on a RPI using socketcan and then simply switch to e.g. stm32. That would also cover the #15 issue as the [identifier](https://docs.rs/embedded-can/0.3.0/embedded_can/enum.Id.html) is handled as an enum. I have it partially implemented [here](https://github.com/marcelbuesing/socketcan-rs/tree/embedded-hal). But if there is a breaking v2.0 planned I would probably also change functions such as `CanFrame::new` to take `embedded_can::Id` as parameter instead of `u32`.

cwoodall, 2021-02-21:
I agree some sort of abstraction that would work cross platform would be great for my needs. Definitely willing to help a little

fpagliughi, 2021-02-21:
@marcelbuesing That does sound like a good idea for a unified CAN package, but I'm not too experienced with using Rust on bare metal, so might not be too helpful getting there. And being that this is the "socketcan" crate, maybe a unified one should be something new that possibly wraps and depends on this one for the Linux implementation?

I think the scope of this work, over the next few weeks, should be to get this library modernized and covering as much of the SocketCAN API as possible, including the Netlink configuration functionality. And it should be a breaking, major version update to v2.0 to include #15. I also like the idea in #10 to make timestamps an (optional) part of the frame, rather than returned separately as a tuple as is in `master` at the moment.

marcelbuesing, 2021-02-22:
> @marcelbuesing That does sound like a good idea for a unified CAN package, but I'm not too experienced with using Rust on bare metal, so might not be too helpful getting there. And being that this [...]

I think for crates in the Rust embedded environment it's usually HW specific crates implement the  `embedded-hal` traits and you pick the crate that fits your HW requirements, rather than a single crate with multiple implementations inside. Which kind of makes the system more distributed and the embedded-hal implementation is pretty stable. I agree though that priority should be to cover most of the socketcan functionality. I think we could have both though. Meaning one will have to exchange some code if using hardware or system specific functions or values. E.g. [embedded_can::CanFrame](https://docs.rs/embedded-can/0.3.0/embedded_can/trait.Frame.html) does not have a function  `timestamp` but a `socketcan::CANFrame` could still have one. I could fork your current dev and make a PR just to show how it could look like.

marcelbuesing, 2021-04-14:
@mbr @fpagliughi would you accept another maintainer? Seems progress has kind of stalled.

fpagliughi, 2021-04-29:
Hey all. Sorry. Had a little mishap right after starting on the project; things broke, doctors called, work missed. All good now, and starting to get back on track.

brainstorm, 2021-08-28:
I'd vouch for @marcelbuesing being one of the maintainers: he fixed an issue I was having in https://github.com/marcelbuesing/socketcan-isotp/pull/1#issuecomment-907588891 switfly.

This repo and the Rust CAN ecosystem would benefit from more maintainers IMHO :)

/cc @fpagliughi @mbr

fpagliughi, 2022-02-14:
Hey all. Apologies, right after agreeing to take on responsibility for this crate, I got ill, missed months of work, and let my open source projects go stale. But now I'm back, feeling good, and catching up.

I have a little bit of un-merged work, but it sounds like several people are way beyond what I started. 
@marcelbuesing ? 
@matoushybl ?

It may be a bit painful to merge a lot of widely diverged branches, but it's probably better for people to put up pull requests, and I can start sifting through them, merging and/or cherry-picking.

What do you all think?

marcelbuesing, 2022-02-15:
Hey @fpagliughi no apologies needed. Great to hear you are doing better. 

I am pretty much open to anything. From my perspective the most interesting parts would be the following:

1. Cleaning up formatting and doing a migration to Rust 2021. I think it would probably make sense to do this first to reduce merge conflicts.
2. Set up a CI for formatting and build checks.
3. Replacing the outdated netlink dependency or completely removing the parts that rely on that, I think you also have that changes in your fork but they probably need to be updated to latest stable neli version.
4. Embedded-hal trait implementation to fix the issue that extended ids described in #15 . This will probably be a breaking change but is required from my perspective as the current ID handling for extended IDs does not make sense.
This would also allow to make the socketcan crate the default implementation for the linux-embedded-hal crate.
6. #16

I will see if I can create a PR for 1 and 2 first. But if anyone finishes this first feel free =).

fpagliughi, 2022-02-15:
Yes, I have a branch that pulled in #13, to switch to the [neli crate](https://crates.io/crates/neli) for Netlink support. The support was broken, and I got a few things fixed, with some help from the neli maintainer(s) and they agreed to help us if we need specific things added to that crate (CAN constants, etc) if we can tell them exactly what we need.

Last night, I had a look, and got it to build with the latest neli, though it's not fully tested yet.

That branch also brings us up to Rust 2018, with std error support.  **I don't think it's time to push to Rust 2021 yet.** 
- Going to 2021 would push up the minimum supported Rust (MSRV), past the one currently supported by the Yocto [Meta-Rust package](https://github.com/meta-rust/meta-rust). So it wouldn't be usable in Yocto builds, which I use.
- I don't think there's anything specific in 2021 that we need yet
- I have a customer using Rust v1.46.0, so for personal reasons, I would want that, at least to be the MSRV. But meta-rust _is_ up to v1.54.0, so I could be convinced to use that.

So I can merge my branch for neli and Rust 2018 within a day, or so, to get started.

@marcelbuesing, If you can do 2 (CI), that would definitely help. I still haven't figured that out yet.

Then we can figure out 4 & 5. 

But I definitely think it's time for a new major version, so breaking changes should be fine and expected.

fpagliughi, 2022-02-15:
Well it appears that `neli` v0.6 doesn't build with Rust v1.46, but does with v1.54. It'll likely be important to keep up with the latest version of that crate if we need changes upstream.

So it seems like v1.54 (Rust Edition 2018) is the best compromise for MSRV - to work with neli and still build with Yocto.

marcelbuesing, 2022-02-16:
> Yes, I have a branch that pulled in #13, to switch to the [neli crate](https://crates.io/crates/neli) for Netlink support. The support was broken, and I got a few things fixed, with some help from [...]

I agree probably no need for 2021 the 2018 is definitely sufficient. I'll look into the CI part.

nnarain, 2022-06-20:
@marcelbuesing @abalmos

Seems like there's some good stuff in progress here just not a consolidated effort. As mentioned above it would be great if there was support for the embedded_hal CAN traits.

@marcelbuesing You mentioned you have a fork with some support for CANopen
@abolmos You mentioned in #20 that you were working on some J1939 support.

I've been dealing with both of these (mostly CANopen) in ROS drivers for the last few months (In C++ not Rust though). I think it would be great if the CANopen and J1939 support was built on embedded_hal that way the embedded world benefits as well.

abalmos, 2022-06-20:
@nnarain I do have a mostly complete rust library for the j1939 kernel module, but never found the time to clean it up and publish it. The semantics felt different enough, I gave up including it as part of this project.

The current version lives [here](https://github.com/oats-center/isoblue/tree/master/services/j1939d/src/j1939) in mostly library form, but embedded in a binary project.

nnarain, 2022-07-07:
MR adding embedded HAL traits: https://github.com/mbr/socketcan-rs/pull/23

This pulls in @fpagliughi updates.

I think it would be nice to at least get https://github.com/mbr/socketcan-rs/pull/20 merged. And updated the on crates.io. Though I do feel the embedded hal traits is a nice edition to a 2.0 version.

fpagliughi, 2022-08-04:
Hey All. I've been waiting to move on some of these updates pending approval by @mbr, as was our original agreement when I started working on the package. He just contacted me to say he doesn't have time to work on this in the near term, so will be giving me full control of this GitHub repo and the associated crates.io. 

So we will be able to start integrating some of the updates and publishing new versions! Very nice. 

Unfortunately, the timing kind of stinks, as I am soon to leave on summer holiday. But I will try to get #20 merged and published as a new baseline, so we can start working around that. Then, when I return in early September, we can get moving on the rest

nnarain, 2022-10-25:
Hey folks. 

I've forked and published a version of this crate call socketcan-hal. This focused on providing an embedded-hal interface to socketcan. 

Happy to open PRs here when ownership is resolved.

See my fork here:

https://github.com/nnarain/socketcan-rs

fpagliughi, 2022-11-23:
OK. Finally! I have some time to get working on this. I verified that I _do_ have write access to this repository, and merged #20 into master. I'll go back through that work and maybe get it cleaned up and make sure that I can publish to crates.io.

Then it's a matter of looking through the existing PR's to see what's useful and can possibly be merged cleanly, then finally gathering all the useful forks that are out there and see what we can get in.

@nnarain let's discuss the embedded-hal stuff on the PR #24.

fpagliughi, 2022-11-29:
Let's all pretend that this is now being maintained! :smile:

closed 2022-11-29 by fpagliughi