socketcan-rs/socketcan-rs #15: Standard/extended identifier implementation incorrect?

stabler, 2020-12-05:
I believe the current implementation of the CANFrame constructor is incorrect, because it does not allow the construction of frames with an extended identifier less than `0x800`.

Code is currently as follows:
```
    pub fn new(id: u32, data: &[u8], rtr: bool, err: bool) -> Result<CanFrame, ConstructionError> {

        ...

        if id > EFF_MASK {
            return Err(ConstructionError::IDTooLarge);
        }

        // set EFF_FLAG on large message
        if id > SFF_MASK {
            _id |= EFF_FLAG;
        }
```

Say I wanted to construct a frame with extended identifier `0x100`. If I pass `id=EFF_FLAG | 0x100` the constructor would return an `IDTooLarge` error. If I pass `id=0x100`, the constructor returns a frame with a standard identifier.

The CAN standard allows for extended identifiers less than `0x800`, and in fact these are commonly used, here are a few examples:
https://stackoverflow.com/questions/63213413/standard-and-extended-messages-with-same-identifier
https://forum.hms-networks.com/t/about-extended-frame-format/2237
https://kb.vector.com/entry/1093/

Recommendation: add an argument to `CANFrame::new` to set the identifier type explicitly (this is a breaking change to the API- but current behavior is incorrect in some cases)

teamplayer3, 2021-11-11:
In the next release of [embedded-hal](https://github.com/rust-embedded/embedded-hal) (0.2.x) the `embedded_can` ID structs exists there as well. I don't know if it's better to use the stand alone `embedded_can` IDs or the other. But using one for most implementations is better for compatibility, I think.

marcelbuesing, 2021-11-11:
I think it would make sense to go with embedded-hal. initially the plan of the embedded-can author was to make it part of embedded-hal and from my understanding embedded-can was basically more of a intermediate step.

tobz, 2022-03-01:
(New user here, have recently started using `socketcan`/`socketcan-isotp`/`embedded-can`/etc...)

It's not clear to me why it would be beneficial to use a crate that is ostensibly designed for providing a HAL for embedded microcontrollers when SocketCAN is generically available for Linux.  Maybe it becomes a forced decision if `embedded-can` goes away and only lives on in `embedded-hal`, but the main types it exposes are so generic to CAN, and so far removed from even SocketCAN or Linux or embedded itself, it's weird/surprising to depend on that crate.

It feels like it could be more beneficial to have a foundational crate -- for example, `can` or `can-core` or whatever -- that defines all of these immutable/fixed/specified types and then other downstream projects can depend on _that_.  Like, the format of a CAN frame, and the meaningful of flags, standard vs extended IDs, ISO-TP... it's already well-defined.  You could likely create a 1.0 release in short order, and avoid worrying about other random tangentially-related crates going away/becoming unmaintained.

fpagliughi, 2022-03-01:
Yes, I've been wondering which is the better way to go on this as well. I suppose there are some microcontroller/RTOS implementations of SocketCAN, like in _Zephyr_:
https://docs.zephyrproject.org/2.6.0/samples/net/sockets/can/README.html

But are there even any Rust ports to any non-Linux embedded systems that use SocketCAN? If not then I probably wouldn't want to spend much time considering designs to incorporate anything that isn't Linux.

That said, if there are generic traits and types for CAN that would be useful to help port between SocketCAN and Embedded CAN crates, then I wouldn't be opposed to using them directly, if they are zero cost. But if they were to impose some performance overhead, I would prefer to isolate them in an optional layer, such as behind a feature flag.

marcelbuesing, 2022-03-01:
I think the [bxcan-crate](https://github.com/stm32-rs/bxcan/) implemented it fairly well. Basically it has it's own copy id types in [id.rs](https://github.com/stm32-rs/bxcan/blob/master/src/id.rs) and a [separate module](https://github.com/stm32-rs/bxcan/blob/master/src/embedded_can.rs) with embedded-hal "translations". The `embedded-hal` part can than still be put behind a feature flag. We definitely need something anyway like the types in `id.rs` to fix currently incorrect handling of extended ids.

I think the point is basically to support different systems easily without relying explicitly on socketcan. E.g. a STM32 embedded controller using bxcan or RPI using socketcan or possibly some commercial windows driver in the future.
Check out the (linux-embedded-hal crate)[https://github.com/rust-embedded/linux-embedded-hal] . I think it would be nice to basically extend this to reference the socketcan crate for the embedded-hal CAN implementation.

The [python can](https://python-can.readthedocs.io/en/master/interfaces.html) project also seems to follow this approach and I personally think it's worth the effort especially for crates that basically build on top of CAN. E.g. imagine a UDS crate (unified diagnostic service not Unix domain socket =) ) that can be used with any of those crates rather than completely relying on socketcan.

tobz, 2022-03-01:
For what it's worth, I (just) acquired ownership of the `can` crate if people are interested in the idea of a foundational crate that could potentially supplant other implementations as the canonical source for stuff like IDs, constants, types, etc.

I think the notion of crates being able to derive their own translations layers based on foundational traits is great, but I don't see a need (personally! there's still a ton I don't actually know about CAN and the CAN ecosystem) for each of those crates to define their own version of standard/extended IDs and then add in a bunch of conversions between other crates.  It really does seem like that should be defined in a single crate.

marcelbuesing, 2022-11-29:
I just ran into this again trying implement the linux-embedded-hal traits. I think this really should be fixed in 2.0 as it pretty likely a breaking change. It's still not possible to create a frame with extended frame flag set but an identifier less than `0x800`. 
https://github.com/socketcan-rs/socketcan-rs/blob/d8e0fb1fbec17578b2394b715427346617516f11/src/lib.rs#L503

Options I see are (in no particular order) :

- add eff: bool parameter to new fn and remove the check in line 503.
- remove both checks in line 498 and 503. I think this is actually a bad idea as it's a hard to find breaking change that people will probably just run into at runtime. Basically you'd very likely end up not setting the EFF once you upgrade to 2.0.
- merge the full version of #24 .
- reduce #24 to just embedded-hal (alpha 8)/embedded-can(alpha 9) ids and replace `id: u32` with `id: Into<Id>`

fpagliughi, 2022-11-29:
Actually, I haven't looked at what the Embedded HAL stuff is doing yet, and I think we will eventually move toward using it, but given the existing library, I would normally just add a second, targeted constructor, like:
```
pub fn new_extended_id(id: u32, data: &[u8], rtr: bool, err: bool) -> Result<CanFrame, ConstructionError> {
  let mut frame = Self::new(id, data, rtr, err)?
  frame._id | EFF_FLAG;
  Ok(frame)
}
```
Or maybe this would be named `with_extended_id()`?

The idea being, keep the base `new()` for the most common cases and make different `new`'s, `with`'s and `from`'s for the more specific cases.

On that note, the most common case would just be to take the `id` and `data` slice, and make the specification of `rtr` and `err` separate.

Or.... make them into a `Flags` type and specify that optionaly:

```
pub fn new<F>(id: u32, data: &[u8], flags: F) -> Result<CanFrame, ConstructionError> 
where
    F: Into<Option<Flags>>
{
  ...
}
```
In this case you can force on the extended ID by specifying it as a flag:
```
let stdFrame = CanFrame::new(0x50, &data, None)?;     // Defaults to standard ID
let extFrame = CanFrame::new(0x900, &data, None)?;    // Extended ID implicit by value

let extFrame = CanFrame::new(0x50, &data, Flags::ExtendedId)?;  // Explicitly specify flag
```

tobz, 2022-11-29:
Circling back, I did end up trying to provide some of these CAN primitives in the aforementioned `can` crate.

I specifically avoided anything related to traits because that can just get too opinionated too fast i.e. sync vs async. If anything in `can` looks appealing to you, or you'd want to take it over and use it as a more foundational crate (sort of how many Rust HTTP crates use `http` for the foundational primitives) I'd be happy to chat and/or transfer ownership.

fpagliughi, 2022-12-07:
Several workarounds were added to fix this. The `CanFrame::new()` and `CanFdFrame::new()` functions were refactored to take an `embedded_can::Id` object which must be a `StandardId` or `ExtendedId` type. When it's an extended ID, it sets the extended bit regardless of the raw ID value.

The new lower-level `init()` function which was based on the previous version of `new()` now takes a bitflag for the bits instead of a buch of bool values. The caller can set the extended bit manually, and if not, only then will it guess based on the ID value.

closed 2022-12-07 by fpagliughi