socketcan-rs/socketcan-rs #22: Anyway to capture the timestamp of a CANFrame?

sprijk, 2022-02-17:


fpagliughi, 2022-02-17:
There was an initial implementation in master:
https://github.com/mbr/socketcan-rs/blob/8e4067b59304fcc5d7733c3febceef38ba0c8120/src/lib.rs#L389-L409

It does what you guessed with the _ioctl_, but I'm investigating. I definitely need a hardware timestamp.

fpagliughi, 2022-03-01:
OK, I did a refresher on network time stamping... (How quickly you forget; I just implemented this in C++ last year!)

The SocketCAN documentation only describes the "easy" way to acquire the timestamp using an _ioctl()_ after getting a frame. It gives you the timestamp of the last received package. That's fraught with all sorts of problems around the race for the time stamp in a pair of calls.

The better way is to enable either the `SO_TIMESTAMP` or `SO_TIMESTAMPING` options on the CAN socket and then use the `recvmsg()` call to get the frame. The response for each frame also includes some metadata in a series of headers that should include the timestamp(s).

That allows you to request a software timestamp, hardware timestamp, or both, on each frame.

The `candump` Linux utility shows how to do this in C. First set the socket option:

https://github.com/linux-can/can-utils/blob/ad250a68dc838824f6e986ac352c8d0fc52cea8e/candump.c#L622-L642

Receive the frame:

https://github.com/linux-can/can-utils/blob/ad250a68dc838824f6e986ac352c8d0fc52cea8e/candump.c#L716-L722

Then parse the headers looking for the time stamp(s):

https://github.com/linux-can/can-utils/blob/ad250a68dc838824f6e986ac352c8d0fc52cea8e/candump.c#L746-L766

The big complication for us is that C/C++ code uses a set of macros to iterate through the headers, `CMSG_FIRSTHDR()`, `CMSG_NXTHDR()`. We'll need to see if any crate like `nix` or `libc` have a way to extract that info already, otherwise we'll have to write it ourselves.

abalmos, 2022-03-01:
@fpagliughi Related to my comment in #20, `nix` does support a fully safe way to pull timestamps from `recvmsg`. I don't think you need my PR for `Timestampns` to work

Something like (only tested with `CAN_J1939`, but should work for `CAN_RAW`)

```rust
  let timestamp = true;
  setsockopt(fd, nix::sys::socket::sockopt::ReceiveTimestampns, &timestamp).unwrap();

  let mut data = vec![...];
  let iov = IoVec::from_mut_slice(&mut self.data);
  let mut cmsg_buffer = cmsg_space!(TimeSpec);
  let flags = MsgFlags::empty();

  let r = recvmsg(fd, &[iov], Some(&mut cmsg_buffer), flags).unwrap(),

  let time = match r.cmsgs().next() {
    Some(ControlMessageOwned::ScmTimestamp(rtime)) => rtime,
    Some(_) => panic!("Unexpected control message"),
    None => panic!("No control message")
  }
```

fpagliughi, 2022-03-01:
Nice! Thanks, @abalmos . I will take that and extend it for (optional) hardware timestamps as well. But that is exactly what I was wondering about.

fpagliughi, 2023-04-04:
OK, so this was the _only_ reason I started hacking on this crate to begin with: because I needed hardware timestamps. And now I'm moving to a big rewrite in v2.0... without the timestamp feature! Sorry. But that will definitely be the goal for a v2.1.

lioriz1, 2024-05-23:
Hi
Sorry for the late join-in.
I'm using `socketcan` crate, reading can messages from blocking socket, similar to the example.
Is there a way to get a timestamp like `can-utils` `candump -L can0`: `(1716466971.108267) can0 1C0BAAAF#000012002D00F158`?

fpagliughi, 2024-05-23:
Rather ironically, that is the _only_ reason I agreed to maintain this crate, and nearly three years and multiple updates later, it is still **not** implemented.

So, apologies, no. It's still not in there yet.

lioriz1, 2024-05-26:
Thanks for the replay @fpagliughi 
Waiting for this one
Maybe I will join the crate, hope it will take less then 3 years

jr1221, 2024-06-18:
Also following this issue.  Unfortunately I am not qualified enough to send code, but I can definitely do high-volume performance testing and whatnot if you need.  Thanks for maintaining!