socketcan-rs/socketcan-rs #26: Cross-platform support

XVilka, 2022-08-16:
Inherently, there is no need to use the kernel driver for this, since at least *some* of the features of this crate could be used also on MacOS or Windows. Using conditional compilation and features it might be possible to make it truly cross-platform

abalmos, 2022-08-16:
SocketCAN itself is a Linux Kernel feature. What features would you expect to be cross-platform?

XVilka, 2022-08-17:
I mostly meant in a combination with canutils crate:
- https://crates.io/crates/canutils
- https://github.com/marcelbuesing/canutils-rs

By the way, since there is a new organization **socketcan-rs**, and canutils-rs look abandoned, @marcelbuesing what do you think about moving your project here as well?

mdegans, 2022-11-11:
> SocketCAN itself is a Linux Kernel feature. What features would you expect to be cross-platform?

It would be nice if it built on other platforms, even if it didn't run. In my case I'm writing a library to parse some specific frames, so for example I have some test code with:

```rust
        let frame = socketcan::CANFrame::new(0x1c0, [0x21], false, false).unwrap();
        let remote = Remote::try_from(&frame).unwrap();
        assert_eq!(remote, Remote::Lock(KeyFob))
```

And then I know it my parser works, even without a Linux kernel. WSL works, but I was hoping for it not to be necessary for this. Seems these are the missing ingredients:

```
error[E0433]: failed to resolve: could not find `unix` in `os`
  --> C:\Users\mdegans\.cargo\registry\src\github.com-1ecc6299db9ec823\socketcan-1.7.0\src\lib.rs:63:14
   |
63 | use std::os::unix::io::{AsRawFd, FromRawFd, IntoRawFd, RawFd};
   |              ^^^^ could not find `unix` in `os`

error[E0433]: failed to resolve: could not find `net` in `nix`
  --> C:\Users\mdegans\.cargo\registry\src\github.com-1ecc6299db9ec823\socketcan-1.7.0\src\lib.rs:64:10
   |
64 | use nix::net::if_::if_nametoindex;
   |          ^^^ could not find `net` in `nix`

error[E0432]: unresolved imports `libc::SOCK_RAW`, `libc::SOL_SOCKET`, `libc::SO_RCVTIMEO`, `libc::SO_SNDTIMEO`, `libc::suseconds_t`, `libc::fcntl`, `libc::F_GETFL`, `libc::F_SETFL`, `libc::O_NONBLOCK`
```

Not a big deal cause WSL, but I figured I'd add to the issue. Thanks for the very useful library!

fpagliughi, 2022-11-29:
I'm not sure if this is the right crate to be made "portable" or cross-platform. It is meant for a very specific CAN implementation. 

That said, perhaps there is a portion of this that can be split out into a `can-protocol` crate, or something like that?  Even so, I'm still not sure this is the right crate for pulling that stuff out either. This doesn't create the wire formats for frames or even does Bosch-style bit timing calculations and things like that. (Things you might want to read back and verify). It just makes API calls to the Linux kernel to do all of that.

fpagliughi, 2022-12-07:
Actually, after making some modifications to the `nix` and `libc` crates this week, I realized that this crate is becoming increasingly dependent on upstream features that are only available in the Linux targets of those crates.

So I'm sorry, but I don't see any of this crate being cross-platform. I'm going to close this.

But look into the Embedded HAL/CAN work. That seems to be forming a base set of CAN definitions that are intended to be portable.