socketcan-rs/socketcan-rs #28: Ideas for v2.0

fpagliughi, 2022-11-29:
Hey All. Thanks for all the feedback recently. It feels good to finally get some inertia on this crate  after so long!

I'd like to get v2.0 released by the end of the year, if that's possible. Here's what I'm thinking for it:

- Get all the outstanding issues and PR's done:
    - @j4r0u53k is working on #16 to bring in CAN FD support.
    - @nnarain 's PR  #24 would solve some of the API and the ExtendedID issues.
    - @marcelbuesing 's #21 would restore CI to the repo

- Make better use of the upstream crates.
    - Some of the low-level types, constants, and other definitions have since been added to the upstream crates, like `nix`. So give a quick review and remove whatever possible from this one.

- [Maybe] Consolidate the error types or make a wrapping `Error` type with a `Result`
    - Lately I've been finding it tough to use a lot of libraries in an application when each has a dozen error types.
    - It's nice when a library has a single, top-level error to say, for example, "Was this a CAN error?"

- Overall cleanup:
    - Better documentation on all the types, functions, modules, etc
    - More restrictive linte like `#![deny(missing_docs)]`, etc
    - Unit Tests!

- Timestamps
    - The whole reason I wanted to update this crate in the first place was because I needed hardware timestamps!  Now I can't even remember if I did anything about that.

After that is finished and released,  I would like to start putting together v2.1 to add full NetLink support for the CAN interfaces.

XVilka, 2023-03-14:
Sorry for the noise, but were there any updates for the 2.0 release? If something isn't finished yet, I still think it's better to craft the new release, so users (us) could use it already, while other improvements can land in next versions.

jreppnow, 2023-03-14:
There was some work on the Netlink portions of this crate around December with a lot left to do, but I also agree that stuff like CAN-FD should be made available as fast as possible (I also need it for other projects).

nnarain, 2023-03-14:
I'd agree we should get a 2.0 released soon. I'm currently still using a fork for tokio-socketcan and would like to put a PR up for that.

fpagliughi, 2023-03-14:
Yes, apologies, I got caught up on several other projects, but I need this soon, too! I had a few experiments started in January on how to clean up and finish some refactoring, but I started wondering recently if I should mostly push out what's been sitting in the repo as v2.0, and then do the further work as v3.0 if it breaks the API again.

But either way, I'll be starting back on this project this week or next at the latest...

therishidesai, 2023-03-21:
Getting 2.0 out sooner will probably also get more people using this crate again, which will help get more people contributing to a cleanup.

fulup-bzh, 2023-04-02:
After browsing every RUST/CANBUS implementations I found on github, I ended up restarting a new can-rustlib from scratch. My only target is Linux and I want to avoid any non critical dependencies. We do LTS support and when running a code for 10/15years you should think twice before adding any new dependency. My is to limit this lib to low-level Can interface, leaving tooling, async,... outside of the library.

I already had a legacy C++ Can service(doc@[1] source@[2]) that supports RawCan+BMC+J1939+OBD2+NMEA2000+... My goal is to renew this service in RUST. This new version should integrate with our new micro-service architecture[3] and VSS/kuksa_databroker[4].

From a functional point of view, Rust/new version should support:
* every previously supported functionalities 
    * RawCan+BMC+J1939+OBD2+NMEA2000+...
    * Microservice per message/signal Security model (Rest,Websocket,Ram/Ram,...)
* New features 
    * faster/less resource 
        * better memory heap usage
        * lazy synchronization with less json serialization
    * message parsing: move from statics OpenXC generating static C++ code to dynamic DBC format interpretation
    * hopefully Rust code will be simpler/cheaper to maintain that C++ version 
After two days of work, I already get promising result:
  * DBC parsing works (thanks to can-dbc [5]).
  * Message binary parsing done with bitvec
  * SocketCan interface done at 80%
     * Raw CAN and CANFD done
     * Filter done
     * Message parsing done
     * BMC in progress (hopefully only an other 1/2day of work) 
I still have to integrate higher level CAN protocol (J1939, NMEA2000, ...) and finalize the API to get a clean separation in between the low-level and higher level.

Hopefully I will be able to publish something by the end of the week. But the code should under "work in progress" status until end of April. 

[1] https://docs.redpesk.bzh/docs/en/master/redpesk-core/canbus/1-Architecture.html
[2] https://github.com/redpesk-common/canbus-binding
[3] https://github.com/redpesk-common/afb-librust
[4] https://github.com/eclipse/kuksa.val/tree/master/kuksa_databroker
[5] https://github.com/marcelbuesing/can-dbc.git

fpagliughi, 2023-04-03:
That's awesome, @fulup-bzh!

As for this crate, the final cleanup of the frame and error types is underway for v2.0, and should be done and released in the next week or two. That work is mostly up in the `develop` branch and should be merged into `master` this week.

Comments welcome.

fpagliughi, 2023-04-04:
I will close this now. If there are any specific problems with the code pertaining to the v2.0 release, please open specific tickets.

closed 2023-04-04 by fpagliughi

osiler, 2023-04-26:
> I'd agree we should get a 2.0 released soon. I'm currently still using a fork for tokio-socketcan and would like to put a PR up for that.

Agreed, what is the best way for moving forward with such a feature flag integration?

fpagliughi, 2023-04-26:
v2.0 is out. If you want to discuss additional features, feel free to open more tickets with ideas. I'd personally like to integrate async/await capabilities into this library, but in a way that is agnostic to the executor framework. Obviously tokio is the most popular, but I have several customers that use others.