socketcan-rs/socketcan-rs #29: socketcan2-rs

xR3b0rn, 2022-12-02:
Since there isn't that much progress in this library for some time, I published my implementation of a libc-socketcan wrapper. You can find it here: https://crates.io/crates/socketcan2

socketcan2 supports timestamps which also work in combination with multithreading (in contrast to the method used in this project) and it is possible to wait on multiple CAN devices in the same thread using one timeout.

Some features are missing right now, such as sending and filtering of frames.

fpagliughi, 2022-12-02:
Actually there's a new maintainer here and a bunch of work currently in progress for a 2.0 update by the end of the year. If you would like to contribute here rather than keep a separate crate, by all means do so. There are a couple PR queued up to be merged in order (#16 then #24), after which new ones will be considered - hopefully in about a week.

But I also have a (thread-safe) update for timestamps that is partially complete and not yet pushed. Maybe we can compare and merge them?

xR3b0rn, 2022-12-02:
Well that's a great idea. Before I create a PR, you can check my repo, if you like my design. I already have some ideas of how to merge my stuff into this project, which we can discuss then.

fpagliughi, 2022-12-02:
In general, it's the same idea to use a timestamping socket option rather than the separate `ioctl()`, noting a few things:

- Enabling timestamps should be optional. Applications should not suffer the overhead of they don't need them.
- I (personally) need hardware timestamps, so would look toward the SO_TIMESTAMPING socket option. Users should be able to select hardware or software timestamps or both.
- I want to move the crate toward more use of `nix` over `libc` when the functionality is available. It's higher-level, less verbose, and usually far less cryptic.
- Timestamps are absolute points in time, _not_ `Duration`'s.

The incoming PR's I mentioned will refactor the existing library to a large extent, as well as the cleanup afterwards. I've been waiting for that to settle before thinking too hard on the API for this. The biggest issue is making it a completely "zero-overhead" feature for applications that don't use it.

But before we start a lengthy discussion here, note that there's an existing issue for adding timestamps: #22. Have a look at that, and then let's continue the conversation there.

xR3b0rn, 2022-12-02:
* I aggree on that
* On that as well
* I don't know `nix` yet, however since the socketcan library is Linux dependent by definition, I love the idea of just forwarding the last OS error
* Well, timestamps are relative to the year 0, so they represent the duration which ellapsed since then. Just think of Unix timestamps... What are they in your definition? Timepoints or Durations? You can't classify them that way. Timestamps always have to be relative to some other point in time, as well as Durations are. That's also why I was surprised that Rust's standard lib durations can't be negative. I mean why can't I have timepoints in the past? Makes no sense to me.

Another important point of my design is the `CanGroup`-class which makes it possible to wait on multiple devices in the same thread, which currently is impossible in this library.

fpagliughi, 2022-12-02:
The `CanGroup` might be a good idea, but I'm not sure that it's worth re-implementing for something like a CANbus library. There are any number of crates that wrap poll/epoll and can be used with structs that implement the `AsRawFd` trait, as the `CanSocket` does. Supporting the grandaddy `mio` crate would be easy enough as well. 

But the bigger assumption is that this crate will support async/await directly, soon enough. That may not be what everyone wants, but given all the options, I'm not sure that writing and maintaining yet another solution in this crate would be worthwhile.

fpagliughi, 2022-12-02:
BTW... here's an example of using the existing library with mio. I blatantly stole it off of stackexchange and modified for socketcan. You can register an arbitrary number of sockets for the event loop. 

```
use std::{
	os::unix::prelude::AsRawFd,
	error::Error,
	io,
};
use mio::{
	unix::SourceFd,
	Events, Interest, Poll, Token,
};
use socketcan::{CanSocket, CanFilter};

const TOKEN_SOCKET: Token = Token(0);

fn main() -> Result<(), Box<dyn Error>> {
    let mut poll = Poll::new()?;
    let mut events = Events::with_capacity(128);

    // Setup our raw file descriptor (using a SocketCAN Socket)
    let sock = CanSocket::open("can0")?;
	let filter = CanFilter::new(0x110, 0x7FF)?;
	sock.set_filters(&[filter])?;

	// This is needed to reads never block indefinitely and halt the loop
    sock.set_nonblocking(true)?;
    
    poll.registry().register(&mut SourceFd(&sock.as_raw_fd()), TOKEN_SOCKET, Interest::READABLE)?;

    // Our event loop
    loop {
        let poll_timeout = None;
        poll.poll(&mut events, poll_timeout)?;

        for event in &events {
            match event.token() {
                TOKEN_SOCKET => {
                    // Here, mio detected that our socket is readable!
                    // we can most likely read events from it!
                    loop {
                        match sock.read_frame() {
                            Ok(frame) => { println!("{:?}", frame); },
                            Err(e) if e.kind() == io::ErrorKind::WouldBlock => break,
                            Err(e) => panic!("Socket error: {}", e),
                        }
                    }
                }
                _ => unimplemented!("other token responses")
            }
        }
    }
}
```

xR3b0rn, 2022-12-02:
NIce thank you. This is obviously better, since it also works with any other kind of file descreiptor such as files, fifos, etc.

xR3b0rn, 2022-12-05:
@fpagliughi couldn't find a way to register on events with user-data in `mio`. Am I missing something? Becuase if it is not possible it's unuseable.

fpagliughi, 2023-04-04:
I never had a chance to look into this. Did you make any progress with it @xR3b0rn ?

closed 2023-05-14 by fpagliughi