socketcan-rs/socketcan-rs #32: Netlink (neli) code for the CAN interfaces

fpagliughi, 2022-12-06:
It would be great to have full coverage of the Netlink API to interact with the CAN interfaces. Or even something better than what is already there. I don't have any experience with the Netlink CAN interfaces, so if anyone has some existing code to get it started, that would be helpful.

@marcelbuesing ? @jreppnow ?

jreppnow, 2022-12-06:
Personally, I have never used it, but I believe we have some sections in a proprietary code base at work that make use of them. If there are any concrete questions, I could possibly cross-reference. However, if the question is "What is even possible?", I'm afraid I can't give a good answer.

fpagliughi, 2022-12-07:
Thanks, @jreppnow . Setting the baud rate and secondary (FD) baud rate would be a great next step.  

The code to bring the interface up and down is in there, and to my knowledge works. It's probably a good example to get started on the rest.

If you were to get started on it, let us know, so that the work is not duplicated.

jreppnow, 2022-12-07:
I'll have some time to look at this in detail over the next weekend, although I'll presumably have to do quite a bit of research first. If that's fine in terms of delay, I'd gladly take a swing.

fpagliughi, 2022-12-07:
Sounds like a plan. Over the weekend I will get back to looking at timestamps.

jreppnow, 2022-12-17:
So I started working on the bitrate stuff in order to use it as an example for a guide, but it's pretty hairy tbh. The main issue is that a lot of constants and structs are missing and it's overall very fiddly. You can see my current progress here: https://github.com/jreppnow/socketcan-rs/tree/netlink-bitrate 

I still going to go ahead and describe my basic approach so that others can follow through. Recommended reading: https://man7.org/linux/man-pages/man7/netlink.7.html and specifically https://man7.org/linux/man-pages/man7/rtnetlink.7.html. 

For cross-referencing, I recommend looking at the iproute2 source code at https://github.com/shemminger/iproute2.

## Basic concepts 

- we are using Netlink sockets to talk to the routing subservice of the Linux kernel, which also handles configuration of network devices
- the message we exchange always have the same format: 
  - Netlink header (in ```neli```, this is ```Nlmsghdr```)
  - Netlink route header (```Iinfomsg```)
  - Netlink route attributes 
    - In general, these are added by ```addattr_l(...)``` and similar calls in the C code. They are put into the ```attributes``` field of ```Iinfomsg``` in ```neli```. 
    - They can be nested  (possibly multiple times)! In ```iproute2```, you will see something like ```addattr_nest(&req.n, sizeof(req), iflatype)``` the code, which opens a nesting scope, and ```addattr_nest_end(...)```, which ends a nesting scope. This is modeled in ```neli``` via the ```add_nested_attribute(...)``` method on the ```Rtattr``` struct. The data content of the ```Rtattr``` that accepts such nested attributes should be Vec<u8> resp. the ```neli::Buffer``` wrapper. 
```rust
        let info = Ifinfomsg::new(
            RtAddrFamily::Unspecified,
            Arphrd::Netrom,
            index.unwrap_or(0) as c_int,
            IffFlags::empty(),
            IffFlags::empty(), // The documentation says this should always be 0xFF..FF, but that does not work!
            {
                let mut buffer = RtBuffer::new();
                /// Adding an attribute.
                buffer.push(Rtattr::new(None, Ifla::Ifname, name)?);
                /// Adding an attribute with nested attributes inside.
                let mut linkinfo = Rtattr::new(None, Ifla::Linkinfo, Vec::<u8>::new())?;
                /// Adding the nested attribute itself.
                linkinfo.add_nested_attribute(&Rtattr::new(None, IflaInfo::Kind, kind)?)?;
                buffer.push(linkinfo);
                buffer
            },
        );
```

## Concrete example 

As mentioned before, I am trying to get bitrate setting to work at the moment. Here is how I approach the problem: 

- Figure out what the correct ```iproute2``` command is, in this case it's 
  ```sh
  ip link set <dev name> type can bitrate <bitrate> [sample-point <sample point>]
  ```
  A good source for these is https://www.kernel.org/doc/Documentation/networking/can.txt. 
- Use ```strace``` to figure out what the command is actually doing: 
  ```sh
  sudo strace ip link set <dev name> type can bitrate <bitrate> [sample-point <sample point>]
  ```
  - This gives us (among other things): 
  ```
  sendmsg(3, { 
    msg_name = { 
      sa_family=AF_NETLINK, 
      nl_pid=0, nl_groups=00000000
    },  
    msg_namelen=12, 
    msg_iov = [ 
      {  
        iov_base= [ 
          {  // This is the netlink header (Nlmsghdr)
            nlmsg_len=84, 
            nlmsg_type=RTM_NEWLINK, 
            nlmsg_flags=NLM_F_REQUEST|NLM_F_ACK, 
            nlmsg_seq=1671295701, 
            nlmsg_pid=0
          }, 
          { // This is the netlink route header (Iinfomsg) 
            ifi_family=AF_UNSPEC, 
            ifi_type=ARPHRD_NETROM, 
[... 75 lines cut ...]
                let timing = can_bittiming {
                    bitrate,
                    sample_point: sample_point.unwrap_or(0) as u32,
                    tq: 0,
                    prop_seg: 0,
                    phase_seg1: 0,
                    phase_seg2: 0,
                    sjw: 0,
                    brp: 0,
                };
                data.add_nested_attribute(&Rtattr::new(None, rt::IflaCan::BitTiming, unsafe {
                    std::slice::from_raw_parts::<'_, u8>(
                        &timing as *const can_bittiming as *const u8,
                        size_of::<can_bittiming>(),
                    )
                })?)?;
                buffer.push(link_info);
                buffer
            },
        );

  ```
  What I am stuck on right now is that IflaCan does not implement the required traits for ```neli``` and adding them is a bit of a pain. Feel free to see the branch yourself. 

I will update this instructions as I obtain more information or based on feedback.

fpagliughi, 2022-12-17:
Thanks for keeping at this and all the updates. My guess is that there's too much upstream work to try to get in before this is workable? So, it sounds like it won't make it in to the upcoming v2.0 release. But it is something we can target for a followup v2.1 release?

jreppnow, 2022-12-17:
Yeah, I think 2.0 should not wait for for the Netlink stuff - making stuff like FD support and the layout fixes available to people seems more important. 

I'll personally keep pecking at this topic, implementing a function/request or two when I have the time. Not sure if we need a specific milestone like a version 2.1 for this tbh.

fpagliughi, 2023-01-04:
Unfortunately, I didn't get much done on this over the holidays, but I'm starting back on it now to get a release out. I began adding some support for the new Netlink features to the optional utility app, which I renamed to `rcan`. This is currently in the `develop`branch. So far, it looks good, but could use some better error messages.

Diving into the Netlink stuff, it definitely looks like we would want to get some stuff added upstream into the `neli` and `libc` crates to support this. Probably the contents of the `can/netlink.h` header:
https://elixir.bootlin.com/linux/latest/source/include/uapi/linux/can/netlink.h

I opened a new issue in the `neli` crate for some guidance on how best to proceed.

fpagliughi, 2023-04-04:
We'll go with what's currently in there for v2.0, and keep this issue open to add some more neli features in v2.1

fpagliughi, 2023-10-11:
Thanks to @jackyzjk  for adding the required structs and enums to implement `set_bitrate()` much the way @jreppnow showed it, above.

I then kept inertia going to implement most of the other netlink CAN structs etc. Just as I was finishing, it occurred to me that I could have just used bindgen, like:
```
$ bindgen /usr/include/linux/can/netlink.h -o bindings_can_netlink.rs
```
But, I did use the output to adjust some of the declarations.

After that I just did some copy-pasta to implement a few more setter functions and commands, like `set_restart_ms()` and `restart()`.

But there are still a few more to go.

fpagliughi, 2023-10-11:
For completeness, I'll add this...

The API definitions for communicating with the kernel about CAN are in the Linux kernel sources here:
[include/uapi/linux/can/netlink.h](https://github.com/torvalds/linux/blob/master/include/uapi/linux/can/netlink.h?ts=4)

And although looking at an existing client like [iproute2](https://github.com/shemminger/iproute2) is quite helpful, it may also be useful to see the kernel code that receives and processes the netlink requests for CAN. The code is fairly small and readable:
[drivers/net/can/dev/netlink.c](https://github.com/torvalds/linux/blob/master/drivers/net/can/dev/netlink.c?ts=4)

I found the top of that file particularly interesting where it maps the requests to the data type expected for each:
https://github.com/torvalds/linux/blob/1c8b86a3799f7e5be903c3f49fcdaee29fd385b5/drivers/net/can/dev/netlink.c#L11-L25

The dev.c file in there has additional implementation of the commands which helps to show the expected state of the interface to process particular commands, and the errors returned if not in that state:
[drivers/net/can/dev/dev.c](https://github.com/torvalds/linux/blob/master/drivers/net/can/dev/dev.c?ts=4)

fpagliughi, 2023-10-12:
A number of important netlink commands are shipping with v3.1, including setting bitrate and FD data bitrate, setting control modes, manually restarting the interface, and setting an automatic restart delay time.

But the implementation is still far from complete, particularly in regard to reading status and parameters back from the interface. As usual, and PR would be appreciated! Hopefully we can get some more of this implemented in v3.2.

fpagliughi, 2023-10-15:
Does anyone know how to extract the CAN-specific parameters out of the `Ifla::Linkinfo` attribute that is returned from the kernel. I added a match in the `nl::CanInterface::details()` function to get the link info, but can't figure out how to parse out the nested attributes.

@jreppnow , did you ever figure this out?

jreppnow, 2023-10-15:
@fpagliughi Can you share the bytes that you are trying to decode? I don't own a physical CAN device (privately) that I can use for testing and getting traces, but presumably the messages should have the same format as the one you use for configuration of bitrate etc. I did something similar (proprietarily) for the rust-netlink project.

fpagliughi, 2023-10-15:
Het @jreppnow . Thanks for the quick reply.

I got completely stuck on this for a day, but I think I'm starting to get it now... When requesting the CAN interface details, the CAN-specific parameters like `can_bittiming`, etc, are in the response message in the `LinkInfo` attribute, like you described above.  I've been trying to figure out how to extract it out, like here:

https://github.com/socketcan-rs/socketcan-rs/blob/d065f837166ec928a396b1f5912b189dd27c127e/src/nl/mod.rs#L531-L543

First I was just trying to figure out _how_ to parse down into nested attributes. 
Then I realized that the attributes are nested a little deeper than I thought.
Then it became obvious that for the final collection, we really do want the `IflaCan` enum that we created pushed up and integrated into `neli` data types, with the proper traits defined, for seamless extraction.  And implementing `FromBytes` for the structs is pretty helpful, too.

I think I may be able to get it this evening. I'll post in the morning if I was able to get it working.

Thanks.

jreppnow, 2023-10-15:
I personally really liked the way the rust-netlink project does it, including the tests which serve as examples and documentation as well: https://github.com/rust-netlink/netlink-packet-route/blob/main/src/rtnl/link/nlas/link_infos.rs#L2672

In general, all of these libraries will have a generic way to parse an attribute (NLA), which should give you the bytes (length + data) and the id, which you need to compare to the relevant constants. And then they have a way to define specific types within the library - the link above does it with enums and a couple of traits (Emitable and NlaParse I think?).

fpagliughi, 2023-10-16:
OK. I got it. There's a really messy initial implementation up in the `develop` branch, and it depends on my fork of `neli` with an upstreamed `IflaCan` enumeration. But at least, for now, I figured out the parsing.

fpagliughi, 2023-10-17:
After the better part of a year, this has progressed enough to close the issue! Thanks to everyone who helped on this, especially @jreppnow  and @jackyzjk. 

With v3.2, most of interface CAN parameters can be set or queried. There are a few minor ones that are missing; those will be added eventually as needed, or just for completeness. 

The low-level constants and struct bindings were kept in this crate to speed up the release, but at some point these will be pushed upstream to the `neli` and `libc` crates.

closed 2023-10-17 by fpagliughi