socketcan-rs/socketcan-rs #36: Are the examples out of date for v1.7.0?

therishidesai, 2023-02-17:
It seems like all the examples don't build for v1.7.0

fpagliughi, 2023-02-17:
Do you mean the examples currently in `master`?

If so, then no, they will not work with v1.7.0. The master branch is preparing for a major update to v2.0. So, relative to that, it's v1.7.0 that is out of date.

therishidesai, 2023-02-17:
```
12 | use socketcan::{CanFrame, CanSocket, Frame, Socket};
   |                 ^^^^^^^^  ^^^^^^^^^  ^^^^^  ^^^^^^ no `Socket` in the root
   |                 |         |          |
   |                 |         |          no `Frame` in the root
   |                 |         no `CanSocket` in the root
   |                 no `CanFrame` in the root
   |
   = help: consider importing this trait instead:
           embedded_can::Frame
help: a similar name exists in the module
   |
12 | use socketcan::{CANFrame, CanSocket, Frame, Socket};
   |                 ~~~~~~~~
help: a similar name exists in the module
   |
12 | use socketcan::{CanFrame, CANSocket, Frame, Socket};
   |                           ~~~~~~~~~
help: a similar name exists in the module
   |
12 | use socketcan::{CanFrame, CanSocket, Frame, socket};
```

The blocking example fails to build with the following. I can rename `CanFrame` and `CanSocket` easily but I'm not sure where `Frame` comes from.

therishidesai, 2023-02-17:
Also it looks like the `CANSocket` type doesn't have a `receive` funciton.

fpagliughi, 2023-02-17:
The examples work with v2.0
The examples **do not** work with v1.7.0

v2.0 is not released yet.

There are a significant number of breaking changes in v2.0

therishidesai, 2023-02-17:
Ah ok, I misread your initial reply.

closed 2023-02-17 by therishidesai

fpagliughi, 2023-02-17:
No worries. I need to get back to this and get v2.0 released. I agree. It's confusing.

therishidesai, 2023-02-17:
Do you have an estimated time as to when 2.0 will be out?