socketcan-rs/socketcan-rs #37: 2.0 ?

fulup-bzh, 2023-04-01:
Your README was updated 4 months ago with a nice promise
```
 this library is currently being updated for a v2.0 release
```
Questions: 
* are you planning to push those new update as a Rust crate 2.0? if then do you already have a target date ?
* I also found https://github.com/xR3b0rn/socketcan2-rs which is a much smaller interface and that does not force a dependence to embedded-hal

Would be cool to have unique SocketCan interface for Linux that pull a minimal set of dependencies.
thank you

fpagliughi, 2023-04-01:
As a coincidence, I started working on the final items to clean up v2.0 this week! So hopefully it will be out soon, but it's competing for time on my nights and weekends, so there's no firm date.

There's already a discussion Issue for v2.0 ideas at #28, if you want to add any comments there. 

There was also a fair amount of discussion about adding the dependence on embedded-hal in PR #24, and the idea to includes it was approved and merged. My thinking is that this crate is meant to eventually give full coverage to all aspects of the Linux SocketCAN (though not necessarily make it portable to other systems that implement the API, like in an RTOS). So this is meant to be a "full" crate, not a minimal one, though I'm completely behind the idea of putting some of it behind feature flags so that unnecessary parts of it can be omitted if you don't need them.

Considering the large dependencies like `nix` and `libc` it's tough to see how a tiny crate like `embedded-can` is a big problem - other than a moving API that is not quite stable yet. But I would have no problem considering a PR that made the dependence optional.

fpagliughi, 2023-04-03:
Conversation continues on #28

closed 2023-04-03 by fpagliughi