socketcan-rs/socketcan-rs #39: Async/Await

nnarain, 2023-04-26:
Hello,

@fpagliughi you mentioned in https://github.com/socketcan-rs/socketcan-rs/issues/28#issuecomment-1523615862 that you were interested in having this library support async/await.

I have a fork of tokio-socketcan using the v2 interface (need to update to the latest from crates.io): https://github.com/nnarain/tokio-socketcan

Personally happy with either approach, figured we could use this issue to discuss.

Firstly, I'm not sure if it's possible to have executor agnostic async/await implementations. Maybe someone can provide insight on how that might be accomplished.

fyi @marcelbuesing @oefd

fpagliughi, 2023-04-27:
What timing. I was just going to open an issue on tokio-socketcan asking if we should update that library and consider merging it with this one.

I was just going to look at it first to get a feel for how it's implemented to figure the best way to make an "agnostic" version. If there's not an easy way to do it at a lower level , then the next best option is to make the runtime selectable through a cargo feature. Then implement it for tokio and maybe one other like async-std or smol. (Although those two might be the same implementation).

nnarain, 2023-04-28:
From the above, seems like feature flags for specific executors might be the way to go

fpagliughi, 2023-05-03:
Agreed. Feature flags. 

The [tokio_socketcan](https://github.com/nnarain/tokio-socketcan) crate looks like a good (and not terribly difficult) way to do it for tokio.

Both async-std and smol wrap the [async-io](https://crates.io/crates/async-io) crate, so I assume one solution would cover both of them.  I'm starting to test it out and it doesn't seem too difficult either.

Then we're done!

nnarain, 2023-05-04:
Awesome. I'll take a look at moving my changes to this crate in the next couple of days.

fpagliughi, 2023-05-05:
I opened an [issue](https://github.com/oefd/tokio-socketcan/issues/9) in the tokio-socketcan crate about merging that repo into this one, and looking at it, I would definitely like permission from the authors before merging. The crate isn't clearly licensed; it says "MIT" in the _Cargo.toml_ file, but nowhere else, although no claim is made for the copyright.

Several of my industrial customers, (and all the biotech/medical) do copyright reviews, so I need to have things in legal order.

jreppnow, 2023-05-06:
@fpagliughi How far have you progessed with the async-io version? I have a basic implementation for my CAN-over-IP tunnel (which uses smol atm) that I could extend to support FD as well an drop it as a PR here, if you want.

fpagliughi, 2023-05-06:
Hey @jreppnow , that would be great. I was just starting to research how to do it and look at some options, so I am not far along at all.

For a separate project, I was also just starting to evaluate @warthog618 's (relatively) new Linux GPIO library, [gpiocdev-rs](https://github.com/warthog618/gpiocdev-rs/tree/master/lib), and like the way async is done there:
- Cargo build features for `async_tokio` and `async_io`
- Default to no async. Opt into one or the other.
- Separate sync and async socket structures (i.e. async adds behavoir in a submodule and doesn't replace or remove anything). 
- Simple source/module structure: `src/async/tokio.rs` and `async_io.rs`
- Should be easy enough to implement around the raw socket handles for either case.

jreppnow, 2023-05-06:
I think there is also some argument to be made for replacing the synchronous methods with async ones entirely, as it avoids the very easy programming mistake of calling synchronous IO calls within an async application. However, the major argument against this is that you might be depending on a library that uses the synchronous version upstream, and spawn there calls/methods with something like tokio's spawn_blocking. This constellation would be impossible if the synchronous structs/methods are replaced entirely. 

One issue is that we can't do the trick with the Socket trait to save some duplication, since async traits are still unstable :( Probably gonna use a declarative macro for that one.

fpagliughi, 2023-05-06:
I think if you were starting a standard networking client from scratch, it would likely be easy to chose tokio, or similar, and just use that for simplicity, Especially if it were a technology that might call for a large number of clients connections.

But the state of Rust async/await is far from perfect. It can make hard things easy, but it also makes easy things hard.

Most of my CANbus projects on Linux monitor a single interface. Some two interfaces. And they only tend to manage a few CAN sensors or controllers. So for that, a simple, synchronous API is optimal. Plus it removes a lot of dependencies and additional choices, which you might not need in small apps.

So I'm definitely in favor of keeping the sync API.

warthog618, 2023-05-07:
> I think there is also some argument to be made for replacing the synchronous methods with async ones entirely, as it avoids the very easy programming mistake of calling synchronous IO calls within an async application.

I did consider wrapping all the synchronous methods, but the flipside is that those calls generally complete faster than you could hand the call off to another thread in userspace, so it is more efficient to call them directly.  And I don't want to go imposing overhead for no reason.

OTOH, if your GPIOs are connected to your SOC via I2C, SPI or whatever, rather than being in the SOC itself, then the calls will take more time as they propagate over the bus.  That may or may not be significant to you, so something you need to be mindful of.

> However, the major argument against this is that you might be depending on a library that uses the synchronous version upstream, and spawn there calls/methods with something like tokio's [...]

Indeed, having a library involve itself in spawning threads made me uncomfortable, and I thought it better to leave it to the app to determine how best to deal with that.

nnarain, 2023-05-09:
It's still using my fork or socketcan-rs. I should be able to update that soon

oefd, 2023-05-09:
Just reiterating what I said elsewhere, but:

As the original author I did intend the code to be MIT licensed, just didn't include an explicit license file. I left the job that had me messing with CAN so I haven't been doing much with it for a while, if you can find some use for the code I'm all for it.

nnarain, 2023-05-09:
@fpagliughi Updated my fork. https://github.com/oefd/tokio-socketcan/pull/10. Figured I'd just put up the PR in the original repo but feel free to merge these however you like.

fpagliughi, 2023-05-10:
OK. The git merge is done, but the overall integration is only half complete. 

I grabbed the master branch from [nnarain/tokio-socketcan](https://github.com/nnarain/tokio-socketcan). What you folk want to do with the PR to the tokio-socketcan crate is up to you all.

The library itself is building with a new `async-tokio` feature, but I have not done any testing. I made some additional updates like removing the `Error` type from the module, since it was made redundant with the one in the top-level.

I haven't done anything with the examples or unit tests, but I hope to at least get the examples done before I quit for the day.

fpagliughi, 2023-05-10:
Now the tokio examples are building and appear to run OK. But  I still haven't tried the unit tests.

fpagliughi, 2023-05-11:
All looking good so far. The `async-io` PR #41 is merged into master, along with some other minor updates I had started in the `develop` branch before all this.

We'll probably want to do some work to get the API of the `tokio` and `async-io` sockets a little closer to that of the sync sockets, but we're off to a nice start.

fpagliughi, 2023-05-13:
I split out the functions from the `Socket` trait that get/set options and filters and put then into a trait named `SocketOptions`, then added an impl for it to all the new async sockets. That makes them more consistent. It's in master.

I also added examples for 'smol' and 'async-std' as copypasta from the tokio examples.

Things are still messy but getting a little better.

fpagliughi, 2023-09-19:
v3.0!

closed 2023-09-19 by fpagliughi