socketcan-rs/socketcan-rs #42: CAN XL support

marcelbuesing, 2023-05-12:
I think it would be interesting to add support for this, although I guess there is little adoption so far.

More info e.g. [here](https://lore.kernel.org/linux-can/20220724074402.117394-1-socketcan@hartkopp.net/).

marcelbuesing, 2023-05-12:
I started in https://github.com/rust-lang/libc/pull/3247

fpagliughi, 2023-05-12:
If it's in Linux SocketCAN, I have no problem adding it to this library, even if it's not very popular yet. Since it's only in a recent kernel, though, it should definitely be put behind a cargo build feature that is off by default.

But that's great that you're pushing the low-level structures into _libc_. That's very helpful and consistent.

fpagliughi, 2023-05-12:
Oh, and do you know of any hardware that supports it?

marcelbuesing, 2023-05-15:
> Oh, and do you know of any hardware that supports it?

Seems like there is nothing one can buy yet (or I did not find it). But it seems to be close:

> The connected nodes used CAN SIC (signal improvement capability) XL transceivers from Bosch, Infineon, NXP, and Texas Instruments.
https://www.can-cia.org/news/press-releases/view/3rd-can-xl-plugfest-and-can-xl-technology-day/2023/5/8/