socketcan-rs/socketcan-rs #46: Creating Error Frames For Test Purposes

sjsimps, 2023-07-10:
Hello,

In socketcan 1.7.0, `CanFrame::new(...)` had parameters as options to indicate that the canframe built was intended as an Error frame. See: https://docs.rs/socketcan/1.7.0/socketcan/struct.CANFrame.html#method.new

In 2.0.0, it appears that we cannot construct `CanErrorFrame`s and that this is intentionally not supported. See: https://docs.rs/socketcan/2.0.0/src/socketcan/frame.rs.html#752-754 

For my application, I want to mock something that returns `CanFrame` data.
I want to be able to create `CanErrorFrame`s to test our error cases.

How would I go about creating a CanErrorFrame for the purposes of mocking in tests? This is currently a requirement for us to move from 1.7.0 to 2.0.0.

sjsimps, 2023-07-10:
For anyone who follows this issue, one way to handle this as follows:
```diff
-                CANFrame::new(your_can_id, your_can_data, false, true).unwrap()

+                let mut cf: libc::can_frame = unsafe { std::mem::zeroed() };
+                cf.can_id = your_can_id | libc::CAN_ERR_FLAG;
+                cf.can_dlc = your_can_dlc;
+                cf.data = your_can_data;
+                CanFrame::from(
+                    socketcan::CanErrorFrame::try_from(cf)
+                    .unwrap(),
+                );
```

fpagliughi, 2023-07-10:
Yeah, obviously my thought was that an application wouldn't normally create Error frames, so there shouldn't be a `new()` function to make new users think that's something they should do.

But obviously being able to create them for tests and mock/simulators would be handy. I was thinking of the way you show as a solution.

But I am also fully prepared to say my decision was stupid and over-protective if there's a general consensus to that effect. I didn't want to make it easy, but perhaps just a big warning in the docs would suffice?

sjsimps, 2023-07-10:
Big warning in the docs would probably get the message across.
Maybe an assertion somewhere to not use it to actually send on the wire if there isn't such an assertion already.

fpagliughi, 2023-09-15:
I'm back from summer holidays and will start working on the next release of this library, which will include this.

fpagliughi, 2023-09-19:
OK... Three ways to make a CanErrorFrame:

- `CanErrorFrame::new()` now works, but is kinda weird because it requires an `Id` type.
- `CanErrorFrame::new_error()` is similar but more intuitive using a raw ID word.
- `From<CanError> for CanErrorFrame` to create an error frame from a `CanError`.

fpagliughi, 2023-09-19:
Released in both versions: v2.1 & v3.0

closed 2023-09-19 by fpagliughi