socketcan-rs/socketcan-rs #48: Socket read and write hangs after a while

Woz4tetra, 2023-09-11:
I'm trying to get this library working to connect to a CAN based motor controller. I have a fully functional implementation based on python-can and now I'm trying to port to rust. I have no issues communicating with the motor controller with the python implementation so I know my hardware is correct.

What I've noticed is if I run my code for ~10 seconds, eventually it will lock up either at an attempt to read or write data. Up until that point, I will receive heartbeat packages from the controller as expected.

Also, I can't seem to get data back from my requests. I've double checked the packets I'm sending are the same as the ones from the python implementation.

What can I do to debug this further?

Woz4tetra, 2023-09-11:
I can't share the whole code base but here's the snippets that are important (also this is running along side r2r, a ROS2 client):

main.rs
```rust
...
#[tokio::main]
async fn main() -> Result<(), Box<dyn std::error::Error>> {
    ...
    // Run the subscriber in one task, printing the messages
    spawner.spawn_local(async move { subscriber.for_each(|msg| {
            println!("Received: {:?}", msg);
            command_cb.lock().unwrap().twist = msg;
            command_cb.lock().unwrap().time = std::time::Instant::now();
            future::ready(())
        }).await })?;

    // Run the publisher in another task
    spawner.spawn_local(async move {
        loop {
            let _elapsed = timer.tick().await.unwrap();
            let tank_command = twist_to_tank_drive_velocities(command.lock().unwrap().twist.clone(), wheel_base_distance);
            let tank_velocities = motor_controller.update(tank_command).await;
            let msg = tank_drive_velocities_to_twist(tank_velocities, wheel_base_distance);
            publisher.publish(&msg).unwrap();
        }
    })?;
...
```

roboteq_motor_controller.rs
```rust

    ...
    async fn send_command(&self, index: u16, subindex: u8, data: u32, data_size: u8) -> Result<(), Error> {
        let num_bytes_unused: u8 = 4 - data_size;
        let msg = SDOPacket::command(self.node_id, num_bytes_unused, index, subindex, data);
        let frame = msg.to_can_frame().unwrap();
        self.socket.write_frame(frame)?.await?;
        Ok(())
    }

    async fn process_requests(&mut self) {
        println!("Reading request");
        let next = self.socket.next().await;
        println!("Got data");
        if next.is_none() {
            return;
        }
        match next.unwrap() {
            Ok(frame) => self.process_frame(SDOPacket::from_can_frame(frame)),
            Err(err) => eprintln!("{}", err),
        }
    }
    ...
```

Here's the full file for the above:
[roboteq_motor_controller.zip](https://github.com/socketcan-rs/socketcan-rs/files/12579868/roboteq_motor_controller.zip)

fpagliughi, 2023-09-13:
Is this using the code from master? With the async `tokio` module?
What target board and/or CAN controller? Kernel version?

I just merged in the async/IO stuff from other authors, and one reason it hasn't been released yet is because I haven't had time to do extensive testing of those modules. Let me know if you have any idea how to narrow it down to a simple test app that recreates the problem to help me recreate it.

Woz4tetra, 2023-09-15:
this is the dependencies section of my Cargo.toml file:
```
[dependencies]
r2r = "0.7.0"
futures = "0.3.15"
tokio = { version = "1", features = ["full"] }
tokio-socketcan = "0.3.1"
bytemuck = "1.13.1"
socketcan = "1.7"
```

Looks like I used version 1.7. I can try it on master.

The target is the Roboteq GBL2660 dual motor controller. I've set it to use the CANOpen protocol. 

Here's my OS info:

```bash
$ uname -r
5.15.0-83-generic
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.6 LTS
Release:	20.04
Codename:	focal
```

I will work on making simple test app.

fpagliughi, 2023-09-15:
TL;DR: The v1.7 line is no longer maintained. There's a released 2.0 with some breaking changes, but if you want async/await, that has been merged into master, but not yet released. My advice: Try your code with master and report any problems. This will soon be released and will be maintained.

-----

The 1.7 line was almost completely rewritten and released as v2.0 about six months ago. After that, the `tokio-socketcan` crate, which was written by a different group of people, was merged into master here with their permission, and a comparable `async_io` module was donated to the project for use with async-std & smol.

No one who wrote the 1.x code is still actively working on the project, so it is effectively unmaintained. 

But going forward, with the async stuff added to this crate you wouldn't need the additional `tokio-socktcan` module. That also had to be updated for the modified v2.0 API here. There were some breaking changes in this library. All told, there was an additional breaking change to the API which will likely mean that the next version has to jump a major version to 3.0.

Any new code should undoubtedly be written to v2.0 or later.

fpagliughi, 2023-09-19:
Version 3.0 is out. It includes the integrated async code. Let me know if the issue still exists in this version.

fpagliughi, 2023-10-12:
Any update on this?

fpagliughi, 2023-10-17:
I'll close this due to inactivity. Fell free to re-open if there's anything new to report.

closed 2023-10-17 by fpagliughi