socketcan-rs/socketcan-rs #49: Add support for custom socket protocol (CAN_RAW)

boon-code, 2023-10-09:
Unfortunately, some platforms use custom values for the protocol in the socket syscall instead of `CAN_RAW`.
Would you agree to add support for this, given I prepare and maintain a patch?
I see three ways of providing support for it:
- Simply providing functions in `CanSocket` allowing to specify the protocol number. This is 100% backwards compatible.
- Adding support for it in the `Socket` trait. This would be incompatible for all existing implementations of `Socket` outside of this crate. This would introduce a new function `open_addr_with_proto(addr: CanAddr, protocol: c_int)` which needs to be implemented while the existing function could be auto provided using `CAN_RAW` as protocol.
- An environment variable could be used to allow to override the socket protocol. This could be disabled per default and be linked to a crate feature.

You are also providing a way to pass in an existing socket, so this isn't strictly necessary (however it would be convenient). This special case could also be implemented in the final application alone.

Would you consider merging such a change? Which one of the variants would you prefer (if any of them sounds acceptable at all)? 
I would be happy to provide an implementation if there is a chance of acceptance ;) Of course, I would be open for other suggestions as well.

fpagliughi, 2023-10-09:
Are you talking about BCM datagram sockets? Or some other protocol for raw sockets? If the latter, can you point me to the platforms you're talking about, so I have an idea what we're talking about?

One of the things I had done going into v2.0 of this crate was to push as much low-level code upstream as possible, mostly by sending PRs to the _libc_ and _nix_ crates. But I also started looking into replacing the ffi/nix/libc socket code with a Rust _Socket_ type. There's no `RawSocket` in std, but a few crates have it. [socket2](https://crates.io/crates/socket2) looks pretty good, and it has moved into the rust-lang GitHub, so that's a good sign.

If that were to happen, this would be easy; make a `From<socket2::Socket>` for the CAN socket classes and we're done.

But I haven't had time to make much progress on that yet, so a quicker solution may be OK in the meantime. I would have no problem _adding_ some function(s) so long as they don't break existing apps, and don't create too much confusion.

But mostly I'm curious about where this would be needed.

boon-code, 2023-10-09:
Hey, sorry, I thought I have seen this before, but actually I am just unfortunate enough to work with a partner that had such an implementation. Basically, you needed to use a different value for `protocol` (in the socket call) to get a raw socket. My assumption is, that they actually implemented their own protocol in the Kernel rather than using the "standard" way to go. I couldn't find any report of anybody else doing such a thing.
I can't tell any details (NDA) but it was a Yocto based build using the NXP Kernel on an iMX.6. But it seems to be their own invention.

Sorry for wasting your time. I prepared a patch for using an environment variable to configure a different number for CAN_RAW behind a build feature (https://github.com/socketcan-rs/socketcan-rs/compare/master...boon-code:socketcan-rs:override_via_environment). But actually, there is a much easier solution using a shared object and LD_PRELOAD and simply re-mapping those values on the flawed platform and not having to change any code.

It is too much of a hassle to maintain such a patch for one flawed platform, especially if you want to replace that code. Sorry again.

closed 2023-10-09 by boon-code

fpagliughi, 2023-10-09:
No worries. It made me take a quick look at some of the tests that I started with _socket2_. 

One of the cool things I like about _socket2_ is that, although it uses enums for things like the domain and protocol, it lets you wrap an arbitrary integer to create an enum value. That lets you use any domain and protocol, even if the library writers didn't add support yet. 

So, for CAN (which isn't directly supported) we can do this:

```
use socket2::{Socket, Domain, Protocol, Type};

let af_can = Domain::from(libc::AF_CAN);
let can_raw = Protocol::from(libc::CAN_RAW);

let sock = Socket::new_raw(af_can, Type::RAW, Some(can_raw))?;
...
```

I like the idea that the library can be extended in ways like this without having to fork it or maintain extra code for specific weird platforms. It would be worth doing, especially if we rework the socket implementation.

boon-code, 2023-10-09:
I first thought it would be a good idea to provide a function for this. But in the end, it would only shift the responsibility to handle such a non conformant platform to the application.
The existing structure (I think) would already allow me to create the socket outside without any change to the library. So the extra function would just be convenience. But still, with the same issue as mentioned above.
The environment variable would be something that needed to be added everywhere. A bit more generic, but still... I only realized it once I wrote it and after I read your feedback.
It doesn't really make sense to handle such a platform issue on either of the two levels. By preloading a library, I can keep the quirk required on just this one platform completely isolated from the application and libraries. Handling a platform topic on platform level.
I was just so used to seeing this quirk added to internal libraries that I immediately thought I would need it added before I could start.

I will start focusing on my side project using this lib. If I see a chance of a contribution of value I will let you know (but maybe give it an extra day of thought 😅) if that's okay.
Thanks again for the quick feedback and for sharing the project in general 🙂

Regarding `socket2`: I found the code in this library quite easy to read and understand. `socket2` seems more focused on cross platform and networking (TCP, UDP) for platforms that don't even support SocketCAN. Of course, I don't know how much effort it actually is to maintain that layer directly in `socketcan`, so it might still make sense to consider it.