socketcan-rs/socketcan-rs #52: J1939 support

tristanguerra, 2023-10-22:
Hi,
I've seen the J1939 conversation come up here several times...  Any chance such an improvement would be considered? Has anyone tried to implement it?
Thanks,

PS: glad to see this project is back in active development... looking very good I must say

fpagliughi, 2023-10-22:
I personally don't use J1939 in my work, and am not really familiar with the details... so I wouldn't be looking to implement it any time soon, myself. 

But we also went through the effort (and made this library somewhat more complex) to implement the types and traits from the [embedded_can](https://crates.io/crates/embedded-can) crate. The idea was to allow other libraries to write the higher-level protocols, like J1939, using this library as the implementation for SocketCAN, and other libraries for other targets.

So, my suggestion would be for someone to write a `can-j1939` crate, using the `embedded_can` traits as much as possible, and make a feature to enable using SocketCAN as one of many possible deployment targets.

tristanguerra, 2023-10-22:
I had the same initial thought, but then things got blurry when, if memory serves me correctly, the Linux 5.4 kernel came out with J1939 support (socketCAN too obviously): https://docs.kernel.org/networking/j1939.html

In other words, what you're proposing is definitely the way to go if you want to use the same library in both embedded systems and Linux, but it's reinventing the wheel if you're only targeting Linux.

Make sense?

fpagliughi, 2023-10-22:
Oh, wow. I had no idea it was in the kernel! Thanks for pointing that out.

Yeah, in that case, it should go in this library. Preferably behind a cargo feature, normally off.

reneherrero, 2023-10-23:
I took a stab at it a while back before you started maintaining this and gave up because:

1. stuff was missing in the libc crate
2. this project wasn't maintained

What I ended up doing is simply writing some C code (vomit) and a thin Rust rapper around it.  Not the most elegant solution, but it worked.

Circumstances change...  I'll take another stab at it when I could find some time

PS: could also be used with NMEA 2000 (boats) which is a super set of J1939

reneherrero, 2023-10-23:
PS2: Mind you I did like your original idea, a lot actually.  It just royally ticks me off to have to pay several hundred bucks for the J1939 specs... that's just wrong.

hate these "yeah, we're open, and we want everyone to use it, but give me a ridiculous amount of $$$ for my PDFs" folks.... ok, I feel much better now... sorry for venting  :)

dlips, 2023-12-10:
Hi,
I started tinkering with the J1939 kernel networking, too. Currently I try to wrap my head around the data structures and behavior of the J1939 stack. I am programming in rust for a while now, but OS level stuff and FFI is new to me. I forked the crate and made a separate j1939 module and can already send and receive data. However, I wonder how to continue because the j1939 stack works a bit different than socketcan because the addressing data is not part of the payload but needs to be handled via sockaddr_can. Secondly, the j1939 can be used either with connect()/read()/write() or with sendto()/recv_from(). I am not sure how to best reflect this in the API.

reneherrero, 2023-12-10:
I've been on it too brother.  Very doable, but VERY HARD NUT TO CRACK.
That J1939 C code is.... Not sure what the politically correct word is...
Let's just say "special".

I'm on my sailboat with sporadic Internet access, that's why I'm not done.
Good news: crack this nut and you'll officially be an FFI black belt (and
make a shit load of money afterwards).

I shit you not.

dlips, 2023-12-11:
I think it is ok-ish, although the documentation could definitely be improved :) I just implemented the Read and Write traits, which work well with a `connect`ed socket. For now this is fine for me. I rather will think about the API for working with PGNs and setting filters etc. 

Do you know if we can access the NAME table from the address table in the kernel stack? I found some older sources, which said no, and also newer ones (but still a few years old) which say they are working on a solution for this.

abalmos, 2023-12-11:
I have not been tracking the discussion in detail, but I wanted to share some J1939 code I wrote for another project. Originally I had thought to bring it here, but I'm not sure the socketcan-rs API is the right one for J1939. The project is not currently focused on CAN, so development stopped at its current (working) state. 

https://github.com/oats-center/isoblue/tree/master/services/j1939d/src/j1939

It is all MIT. Fell free to borrow as you see fit.

dlips, 2023-12-11:
Thanks for sharing this. Will definitely have a look at it. I am also not sure if it should be a behind a feature flag in socketcan-rs or be a separate crate. On one hand it’s conceptionally close the socketcan, on the other hand the kernel stack api works a bit differently and I don’t see atm that much code can be shared between the socketcan part in the crate and the potential j1939 part

fpagliughi, 2023-12-11:
Like I said, I've never worked with J1939, so can't be sure whether this is the correct crate for it or not. But a quick look at the code @abalmos just posted, I think may not? Or, at least not yet.

It appears that J1939 over SocketCAN uses dgram sockets? Whereas this library is based on all raw sockets and fixed CAN frames (standard or FD, etc).

But even if this is not the place for the J1939 code, perhaps I could add a few things so this crate could be useful to something built on top of it. For example, I willfully ignored the J1939 and ISOTP fields in the CAN address object. I could certainly add those so that you could use the `CanAddr` type from here, already wrapped up nicely.

And, to be honest, I allowed myself to be talked into having a separate Socket type for each different supported Frame. That made supporting async more difficult and will be unsustainable as we consider adding, integrating, or just supporting higher-level protocols. Maybe we do need a single, base Socket implementation that is more flexible.

dlips, 2023-12-12:
Yes, J1939 uses DGRAM sockets. 

Yes, extending the `CanAddr` to include the additional fields for j1939 would be good, unless this would interfere with the normal socketcan code. I guess the additional fields are just ignored, if someone sets them by mistake and then uses the address for a raw socket? 

In case anybody is interested [here](https://github.com/dlips/socketcan-rs/tree/7a8bb2dfc86eba6ee2f9b4d25e0e2a88df8b2add/src/j1939) is the current state to my forked repo.

dlips, 2023-12-28:
Now during the holiday break, I have some time to continue working on the J1939 support. The basic implementation is [finished](https://github.com/dlips/socketcan-rs/tree/j1939_support). It still needs a lot of polishing. Feedback is very welcome. Here some thoughts:
- For error handling, I currently just pass through the IoErrors from the socket2 crate. Is this ok? I understand that the CanSocket needs its custom CanError because of the conversion of CAN error frames to proper rust errors.
- I created types for PGNs and NAMEs. Would be nice to have a common base for these types similar to the embedded_can Frame and Ids. I looked on crates.io but could not find anything.

fpagliughi, 2023-12-28:
>  I guess the additional [SocketCan J1939] fields are just ignored, if someone sets them by mistake and then uses the address for a raw socket?

Yeah, even with the existing API you can manually create a `libc::can_addr` and convert it into a `CanAddr`, which would technically allow someone to populate the J1939 fields and try to use them to create a Raw socket. I don't know if they're ignored or if you get an error from the kernel. I'll let someone else experiment. :-)

> I currently just pass through the IoErrors from the socket2 crate. Is this ok?

Sure. That's what the `Socket` classes already do. The library's top-level error is either a std I/O error or a parsed failure from an error frame (i.e. `CanError). So returning ether of those would be compatible. We used a local type alias for the I/O errors:
https://github.com/socketcan-rs/socketcan-rs/blob/7851a509bfb077a1722e0bc013ca97d2aeb15252/src/errors.rs#L93-L94

> I created types for PGNs and NAMEs. Would be nice to have a common base for these types

Maybe open an Issue PR in the embedded_can project to get some feedback? See what they think?

dlips, 2024-09-13:
I saw that you added support for J1939 and ISO-TP to the `CanAddr`. I have a working rust wrapper for the J1939 networking for a while now (including async support for tokio), but I developed it in a separate crate. Now, I copied over my implementation to the `j1939_support` branch in my fork: https://github.com/dlips/socketcan-rs/tree/j1939_support

I still need to make some changes before I can create a PR:
- Use the common `CanAddr` type. I made my own but this is obsolete with the changes you made on the `isotp_j1939` branch
- Use the `SocketOptions`. Made a derivate from this crate, which now needs to be merged back 
- Create some examples

fpagliughi, 2024-09-14:
Yes, I started, but got distracted by some other things. So keep in mind that what is there is perhaps not complete and certainly not tested well. As I mentioned above, my intent is to at least cover the low-level basics. So, for example, since we export `CanAddr`, we should at least cover everything that is in the OS struct, like J1939 and ISO-TP support.

So whether it's best to keep the higher-level protocols in separate crates or merge them in here is still to be decided. My thinking leans more towards making separate higher-level crates that are based more on traits than a specific low-level implementation might be better.  That way something like J1939 could perhaps be ported to other platforms; not just Linux/SocketCAN.

olerem, 2024-10-12:
Hi all,

I'm maintainer of kernel net/can/j1939 stack and glad to see you are using it and integrating it to this project. If you have questions, feel free to ask, i'll do my best. If there are wishes or suggestions for documentation, you can help me to extended and make it better :)

olerem, 2024-10-13:
In case you need it, here is documetation update:

https://lore.kernel.org/all/20241013181715.3488980-1-o.rempel@pengutronix.de/

dlips, 2024-10-20:
Hi @olerem,
thank you for your offer. I guess it is best to ask the questions on your mailing list? The documentation update you linked is very nice. It clarifies some of my questions, especially regarding the possible returned error codes. Currently, I just pass the errors through as general `IoError`s but it would be better to have more specific errors to allow the user to handle errors more precisely. I wasn't aware of the `SO_J1939_ERRQUEUE` and currently don't understand it fully, but it seems to be useful to allow for more finegrained feedback to the user.

olerem, 2024-10-20:
Yes, the preferred way is to ask over linux-can@vger.kernel.org mailing list. I'll recommend to use SO_J1939_ERRQUEUE in most implementations.

fpagliughi, 2024-10-24:
@olerem, thanks for checking in!