socketcan-rs/socketcan-rs #56: Version 3.3.0 Not Compiling With Neli

Peyton-McKee, 2023-12-01:
Hi SocketCan Team, Im working on an embedded project and am trying to build socket can for the newer versions, however any version above 2.0.0 fails to compile when I cargo build due to what seems to be a conflict with the neli library. 

My Cargo.toml looks like this 
```[package]
name = "calypso"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
socketcan = "3.3.0"
paho-mqtt = "0.12.3"
nalgebra = "0.32.3"
```

When I move socketcan down to version 1.2.0 it compiles and works fine, but I want to upgrade to 3.3.0 so that my cargo audit passes.

Heres some sample Terminal output from my cargo build command. 
[image]

Any advice would be much appreciated

fpagliughi, 2023-12-01:
Much of the work for versions 3.1 and 3.2 were to add additional Netlink support to the crate using neli, so things were working well a few weeks ago.

What platform are you targeting?

Peyton-McKee, 2023-12-01:
I’m currently developing on mac, targeting a raspberry pi however its not compiling on my machine, I havn’t tested it on the pi.

fpagliughi, 2023-12-01:
Ummm... does a Mac even have SocketCAN? Or Netlink? I thought those were pretty Linux-specific subsystems. I have no plans of trying to support this on any other OS than Linux.

Peyton-McKee, 2023-12-01:
Hmm, yeah I’m not trying to run the program on the mac, just trying to build it. The actual program runs solely on raspberry pis. Maybe I should just work in a dev environment on docker.

fpagliughi, 2023-12-01:
Yeah, unless you're specifically targeting a Linux system, you won't have the correct components from the upstream libraries.

Have you tried `cross`? I haven't used it on this project specifically, but I do use it a lot when developing the Paho Rust stuff. It's pretty nice.

Peyton-McKee, 2023-12-01:
I have not, havn’t heard of it actually. Ill look into that. This is the first rust program I’ve created so still learning all the rusty features.

fpagliughi, 2023-12-01:
Yeah, cross is a container system for cross-compiling rust. Basically wraps and acts like cargo.
https://github.com/cross-rs/cross

If you already have docker installed, it can use that. Otherwise it can use Podman, I think.

closed 2023-12-01 by Peyton-McKee

baxterjo, 2024-01-11:
> Perhaps I should put in a _build.rs_ file which fails with a clear, distinct message on non-Linux targets.

That could be helpful, maybe add a note that refers to cross compilation for development or even just refers to this ticket.