socketcan-rs/socketcan-rs #57: Tokio-Async CanFd not capable of sending RemoteFrame?

AsimmHirani, 2024-01-31:
I noticed that the Sync and async-io implementations of write_frame followed the CanAnyFrame usage pattern for the CanFdSocket, however the implementation in tokio.rs was only capable of sending CanFdFrames. Is there a reason for this API difference for Tokio vs async-io and sync?

Sync write_frame:
https://github.com/socketcan-rs/socketcan-rs/blob/master/src/socket.rs#L648

async-io write_frame:
https://github.com/socketcan-rs/socketcan-rs/blob/master/src/async_io.rs#L90

tokio write_frame:
https://github.com/socketcan-rs/socketcan-rs/blob/master/src/tokio.rs#L163

This is impacting a project where we are trying to send RemoteFrames over a CanFdSocket asynchronously using tokio. Do you have any information about this usecase?

EliteTK, 2024-03-15:
I also ran into this. Sounds like fixing this would require a breaking change.

fpagliughi, 2024-11-15:
Sorry for the slow response. This library is really a group effort by a lot of contributors. I took over maintaining it a while back but only use a small subset of it on occasion. So inconsistencies pop up and I rely on all you folks to let me know. I'll test out the PR and get it in with the next release if it looks good.

PR #69 (2024-12-24): Support sending can frames over tokio canfd sockets (Resolves #57)
The first patch is unrelated and is just a cleanup of a bunch of extraneous path components. I couldn't compile without removing these first.

I think this is probably an acceptable solution. This also fixes #57.

Please let me know what you think. I can probably test this on real hardware soon but the new tests pass (although they currently seem to omit actually verifying that the received frame is of the right type and content etc).
commit 44cff8b5 (2024-12-24): Merge pull request #69 from EliteTK/tokio-canfd-can

Support sending can frames over tokio canfd sockets (Resolves #57)
closed 2024-12-24 by fpagliughi