socketcan-rs/socketcan-rs #70: ISO-TP support

TheButlah, 2024-06-06:
Will ISO-TP support be added?

Would a PR to add support be accepted?

fpagliughi, 2024-06-07:
I personally don't use it, so didn't have any immediate plans to add it.

What would it take to add it? I know, for example, that there are fields in `sockaddr_can` for J1939 and ISO-TP that aren't yet implemented. We should definitely add that level of support, and coordinate it with any effort for J1939 in #52 .

But after that, what needs to be done? I see a vague line or two in the doc:
> An application wishing to communicate using a specific transport protocol, e.g. ISO-TP, just selects that protocol when opening the socket, and then can read and write application data byte streams, without having to deal with CAN-IDs, frames, etc.

But what exactly does that mean?

XVilka, 2024-06-24:
There exists already a third-party `socketcan-isotp` crate, maybe you could cooperate with them somehow: https://github.com/marcelbuesing/socketcan-isotp 

cc @marcelbuesing

wallem89, 2026-03-15:
Maybe someone wants to pick-up this work together with me? I need ISO-TP for Unified Diagnostics Services (UDS) over CAN and I would love to use socketcan-rs instead of another crate

@TheButlah or @marcelbuesing or XVilka?