socketcan-rs/socketcan-rs #71: Fetch stats or stats64 with socketcan

EIC-JayGanatra, 2024-07-18:
I am a beginner in Rust and working with the socketcan-rs crate. I am trying to fetch the statistics available inside the /sys/class/net/<interface>/statistics/ directory. However, I haven't been able to find a way to do this.

I also attempted to combine the netlink_packet_route crate with socketcan, but I encountered an "Address is already in use" error.

Could you please let me know if there is support for fetching these statistics using socketcan? Alternatively, is there a way to successfully combine both crates?

Thank you for your assistance.

fpagliughi, 2024-07-22:
This SocketCAN library uses the [neli](https://crates.io/crates/neli) crate for Netlink functionality. It is enabled with the `netlink` cargo build feature, which is turned on by default.

The code for the netlink stuff is primarily found here:
https://github.com/socketcan-rs/socketcan-rs/blob/master/src/nl/mod.rs

I don't think there's anything in there to query for statistics on the interface, but I agree that it would be nice to have. Do you think you would want to try adding it and sending a PR?

As for the "Address already in use" error, keep in mind that much of the Netlink access requires superuser privileges. So you may need to run as root to get it to work.  In our Netlink module, the functions that require root access are commented as such, like:
```
    /// Sets the data bit timing params for the interface
    ///
    /// PRIVILEGED: This requires root privilege.
    ///
```

fpagliughi, 2026-09-05:
The netlink capabilities were expanded considerably in v4 and generic get/set params functionality was added for anything the library missed.

If there's anything specific that should be added for v4.1 or beyond, open a new ticket with specifics.

closed 2026-09-05 by fpagliughi