socketcan-rs/socketcan-rs #74: CanFDFrames with ExtendedID are not correctly parsed by socketcan::dump::Reader

slenius, 2024-08-11:
I found that candump log messages for CanFD frames with extended ID seemed to have their ID components truncated.

In the following test, the packet has an extended ID of `0x12345678`, however when passed through a dump::Reader it returns a frame with ID `0x678`, and the `.is_extended()` method returns `False`

```
#[test]
fn test_extended_id_fd_can_dump_record() {
    let candump_string =
        String::from("(1234.567890) testcan0 12345678##500112233445566778899AABBCCDDEEFF");

    let mut reader = socketcan::dump::Reader::from_reader(candump_string.as_bytes());
    let rec = reader.next_record().unwrap().unwrap();
    let frame = rec.frame.clone();

    let inner_frame = match frame {
        CanAnyFrame::Fd(f) => f,
        _ => panic!(),
    };

    assert_eq!(true, inner_frame.is_extended()); // fails
    assert_eq!(0x12345678, inner_frame.raw_id()); // fails with id=0x678
}```

fpagliughi, 2024-12-24:
Thanks for reporting this. I'll get it fixed in the upcoming release.

fpagliughi, 2024-12-24:
Fix is in master with the unit test which is now passing.

fpagliughi, 2024-12-26:
Released with v3.4.0

closed 2024-12-26 by fpagliughi