socketcan-rs/socketcan-rs #78: Recv receives CANFD and triggers panic

0xsmirk, 2024-10-23:
This is code:
```rust
use socketcan::{CanFdSocket, Socket};
use std::error::Error;

fn main() -> Result<(), Box<dyn Error>> {
    // Open the CAN device
    let socket = CanFdSocket::open("can0")?;

    // Set non-blocking mode
    socket.set_nonblocking(true).map_err(|e| {
        eprintln!("Failed to set non-blocking mode: {}", e);
        e
    })?;

    loop {
        // Read CAN frame
        match socket.read_frame() {
            Ok(frame) => {
                // Print the received CAN frame
                println!("Received frame: {:?}", frame);
            }
            Err(e) if e.kind() == std::io::ErrorKind::WouldBlock => {
                // If there are no readable frames, continue the loop
                continue;
            }
            Err(e) => {
                // Handle other errors
                dbg!("Error reading frame: {}", e);
            }
        }
    }
}
```
When running, the following is the situation
```shell
smirk@veh:~/project/canfuzz$ cargo run 
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.10s
warning: the following packages contain code that will be rejected by a future version of Rust: bitf
warning: the following packages contain code that will be rejected by a future version of Rust: bitflags v0.4.0
note: to see what the problems were, use the option `--future-incompat-report`, or run `cargo report
note: to see what the problems were, use the option `--future-incompat-report`, or run `cargo report future-incompatibilities --id 1`
     Running `target/debug/canfuzz`
Received frame: Fd(CanFdFrame { 45##5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Received frame: Fd(CanFdFrame { 45##5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
thread 'main' panicked at core/src/panicking.rs:221:5:
unsafe precondition(s) violated: ptr::copy_nonoverlapping requires that both pointer arguments are a
unsafe precondition(s) violated: ptr::copy_nonoverlapping requires that both pointer arguments are aligned and non-null and the specified memory ranges do not overlap
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
thread caused non-unwinding panic. aborting.
Aborted (core dumped)
```
The panic triggering time is not fixed.
I want to achieve the same functionality as candump, but use candump without interruption, and write my own code to trigger panic

fpagliughi, 2024-10-24:
I'm just looking at how to recreate this... Just coming in on one ID? What rate? Does that matter?

0xsmirk, 2024-10-25:
> I'm just looking at how to recreate this... Just coming in on one ID? What rate? Does that matter?

The baud rate I set is as follows：sudo ip link set can0 up type can bitrate 500000 sample-point 0.75 dbitrate 2000000 dsample-point 0.8 fd on.
I wrote the following code in the virtual machine based on the hardware PEAK CAN

showier-drastic, 2024-10-29:
Obviously another usage of `copy_nonoverlapping` in the same file has the same problem.

fpagliughi, 2024-10-30:
Agreed, this sounds like a stupid bug. I'll have a look ASAP, and push a new release.

fpagliughi, 2024-11-09:
Sorry for the delay. The fix is released.

But the worst part is that the code for the next release (v3.4.0) sitting in master has a re-implementation and fix that's been sitting there since last March!

While my head is in this, I should probably look to get some of the unreleased code out the door...

closed 2024-11-09 by fpagliughi