socketcan-rs/socketcan-rs #81: Is explicit mio dependency necessary?

jauhien, 2025-02-17:
It seems that at the moment `socketcan-rs` does not use `mio` directly, so this dependency in `Cargo.toml` is not needed (and can be removed from the `tokio` feature as well).

fpagliughi, 2025-02-18:
Hmmm... dunno. This came in when we merged the `tokio-socketcan` crate into this one. There have been a couple of updates to it since then, so maybe someone removed the code dependency and didn't take it out of the toml file.

It doesn't appear to be referenced in the code anywhere, and builds fine on without it. I can remove it.

fpagliughi, 2025-12-12:
Thanks, @jauhien & @dlips.

I need to do a little work on this library and will get it cleaned up. I have also started using [cargo-machete](https://crates.io/crates/cargo-machete), to help find unused dependencies in my projects. Works pretty well.

fpagliughi, 2026-09-04:
Going out with 4.0

closed 2026-09-04 by fpagliughi